adafruit/Adafruit_TinyUSB_Arduino #418: Compiler states: SPI header is missing, causing compilation to terminate.

ZZ-Cat, 2024-05-07:
### Operating System

Linux

### Arduino IDE version

PlatformIO Core v6.1.15

### Board

Adafruit Metro M4

### ArduinoCore version

platformio/atmelsam @ 8.2.1

### TinyUSB Library version

v2.2.3 (all the way up to the latest version)

### Sketch as ATTACHED TXT

**Test sketch:** [tinyusb_bug_test.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/15231586/tinyusb_bug_test.txt)
**PlatformIO Config file:** [platformio.ini.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/15231855/platformio.ini.txt)

### Compiled Log as ATTACHED TXT

**Compiler output:** [build_log.txt](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/files/15231638/build_log.txt)

### What happened ?

The built-in TinyUSB is pinned to v1.9.4, which is outdated for my needs.  
I want the latest version of TinyUSB in my firmware; I started with the very latest version, then found the aforementioned compiler error.  
I tracked it all the way back to v2.2.3. Everything from v2.2.2 and earlier compiles (and flashes) with impunity.

### How to reproduce ?

**In a fresh instance of PlatformIO:**
1. Copy the contents of `platformio.ini.txt` file to `platformio.ini`.
2. Copy the contents of `tinyusb_bug_test.txt` to a `tinyusb_bug_test.cpp` file in the default `src` folder.
3. Build the firmware with `pio run`.

### Debug Log

Not applicable

### Screenshots

Compiler output from another project of mine that uses TinyUSB:  
[image]

As a temporary workaround  for my project, I have pinned TinyUSB to v2.2.2:  
[image]

As observed here, v2.2.2 builds successfully:
[image]

hathach, 2024-05-29:
did the same version setup (core BSP + library) work with Arduino IDE ?

ClutchplateDude, 2024-09-12:
I am also running into this. Will this be fixed?