adafruit/Adafruit_TinyUSB_Arduino #420: "midi_test" does not compile for ESP32S3

jhsa, 2024-05-21:
### Operating System

Linux

### Arduino IDE version

Arduino IDE 2.3.2

### Board

ESP32S3 Dev Module

### ArduinoCore version

2.0.13

### TinyUSB Library version

Latest 3.1.4

### Sketch as ATTACHED TXT

Examples/MIDI/midi_test

### Compiled Log as ATTACHED TXT

`
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:109:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = cdcd_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:109:29: error: 'cdcd_deinit' undeclared here (not in a function); did you mean 'dcd_deinit'?
         .deinit           = cdcd_deinit,
                             ^~~~~~~~~~~
                             dcd_deinit
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:122:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = NULL,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:135:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = hidd_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:135:29: error: 'hidd_deinit' undeclared here (not in a function); did you mean 'hidd_init'?
         .deinit           = hidd_deinit,
                             ^~~~~~~~~~~
                             hidd_init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:161:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = videod_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:161:29: error: 'videod_deinit' undeclared here (not in a function); did you mean 'videod_init'?
         .deinit           = videod_deinit,
                             ^~~~~~~~~~~~~
                             videod_init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:174:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = midid_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:174:29: error: 'midid_deinit' undeclared here (not in a function); did you mean 'midid_init'?
         .deinit           = midid_deinit,
                             ^~~~~~~~~~~~
                             midid_init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:187:10: error: 'usbd_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         .deinit           = vendord_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\device\usbd.c:187:29: error: 'vendord_deinit' undeclared here (not in a function); did you mean 'vendord_init'?
         .deinit           = vendord_deinit,
                             ^~~~~~~~~~~~~~
[... 66 lines cut ...]
         .deinit     = hub_deinit,
          ^~~~~~
          init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c: In function 'tuh_deinit':
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:459:29: error: 'usbh_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
       if (driver && driver->deinit) {
                             ^~~~~~
                             init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:461:17: error: 'usbh_class_driver_t' {aka 'const struct <anonymous>'} has no member named 'deinit'; did you mean 'init'?
         driver->deinit();
                 ^~~~~~
                 init
c:\Users\Joao\Documents\Arduino\libraries\Adafruit_TinyUSB_Library\src\host\usbh.c:461:9: error: called object is not a function or function pointer
         driver->deinit();
         ^~~~~~

`

### What happened ?

It doesn't compile for ESP32S3. Didn't test other boards yet. error attached above

### How to reproduce ?

Compile sketch

jhsa, 2024-05-21:
It doesn't compile for the ESP32S2 also. Just checked.

esp32beans, 2024-05-25:
Reverting to Adafruit_TinyUSB verion 3.1.3 eliminates the compiler errors.

Board: ESP32S3 Dev Module
Arduino IDE 1.8.19
arduino-esp32 BSP 2.0.17

jhsa, 2024-05-25:
> Reverting to Adafruit_TinyUSB verion 3.1.3 eliminates the compiler errors.
> 
> Board: ESP32S3 Dev Module Arduino IDE 1.8.19 arduino-esp32 BSP 2.0.17

Thank you. So it is really a bug.
I will try reverting..

jhsa, 2024-05-26:
3.1.3 compiles..

jhaimerl1, 2024-07-22:
I can't get the example to compile with any version after 2.40 for a Nano ESP32...

DonWT, 2024-09-04:
Still fails (miserably and resoundingly) to compile with version 3.3.3

jhaimerl1, 2024-09-04:
Too bad… using Teensy right now

DonWT, 2024-09-04:
I am using an ESP32 because I want to use both USB and BLE midi.  I will
continue with 2.40 for now although Espressif are readying their own
version of the Tiny USB library.

I think that this looks really bad for Adafruit as this is an essential
library for ESP32 S3 boards.

I might just take a copy of the 2.40 source and remove the Tiny USB library
from my IDE.

hathach, 2025-05-21:
not reproducible with arduino-esp32 v3.2.0 and tinyusb 3.5.0, please update your setup and try again

closed 2025-05-21 by hathach