adafruit/Adafruit_TinyUSB_Arduino #426: 3.1.3 or 3.1.4 causes delay in enumeration

amowry, 2024-06-03:
### Operating System

Windows 11

### Arduino IDE version

2.3.2

### Board

ItsyBitsy NRF52840 Express

### ArduinoCore version

1.6.1

### TinyUSB Library version

3.1.3

### Sketch as ATTACHED TXT

https://github.com/amowry/WARBL2/tree/main

### Compiled Log as ATTACHED TXT

[Compile_Log.txt](https://github.com/user-attachments/files/15536507/Compile_Log.txt)

### What happened ?

I am using a custom NRF52840 USB MIDI device with the ItsyBitsy NRF52840 BSP. Upgrading to v. 1.6.1 of the BSP results in long USB enumeration times of up to several seconds. I have determined that that the cause is changes made to v. 3.1.3 of TinyUSB (v. 3.1.1 works fine). Unfortunately this is difficult to reproduce because if I use a minimal USB MIDI sketch it enumerates correctly. My code is in the link above.

### How to reproduce ?

- Install above code (if you'd like I can provide a UF2 file) for a USB MIDI device.
- The device takes several seconds to enumerate

hathach, 2025-05-21:
please update your setup and try again with v3.5.0

hathach, 2025-05-22:
there is no need for bsp update, you can install latest lib version. Arduino IDE should be able to pick that up instead of bundled version

closed 2025-05-22 by hathach