adafruit/Adafruit_TinyUSB_Arduino #432: ESP32-S3: Adafruit_TinyUSB sending garbage / loopback-like traffic on reading only sketch.

DatanoiseTV, 2024-06-29:
### Operating System

Linux

### Arduino IDE version

2.3.2

### Board

ESP32-S3 Dev Kit

### ArduinoCore version

3.0.2

### TinyUSB Library version

3.3.1

### Sketch as ATTACHED TXT

```cpp
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This sketch is enumerated as USB MIDI device. 
 * Following library is required
 * - MIDI Library by Forty Seven Effects
 *   https://github.com/FortySevenEffects/arduino_midi_library
 */

#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

// USB MIDI object
Adafruit_USBD_MIDI usb_midi;

// Create a new instance of the Arduino MIDI Library,
// and attach usb_midi as the transport.
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

// Variable that holds the current position in the sequence.
uint32_t position = 0;

// Store example melody as an array of note values
byte note_sequence[] = {
    74, 78, 81, 86, 90, 93, 98, 102, 57, 61, 66, 69, 73, 78, 81, 85, 88, 92, 97, 100, 97, 92, 88, 85, 81, 78,
    74, 69, 66, 62, 57, 62, 66, 69, 74, 78, 81, 86, 90, 93, 97, 102, 97, 93, 90, 85, 81, 78, 73, 68, 64, 61,
    56, 61, 64, 68, 74, 78, 81, 86, 90, 93, 98, 102
};

void setup() {

  TinyUSBDevice.setID(0x2E8A, 0x104B);

  // Manual begin() is required on core without built-in support e.g. mbed rp2040
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

[... 294 lines cut ...]
Writing at 0x0002c817... (33 %)
Writing at 0x000323df... (41 %)
Writing at 0x00037563... (50 %)
Writing at 0x0003cda2... (58 %)
Writing at 0x0004256e... (66 %)
Writing at 0x00047a38... (75 %)
Writing at 0x0004cfd6... (83 %)
Writing at 0x000528c1... (91 %)
Writing at 0x0005bdf9... (100 %)
Wrote 339760 bytes (196150 compressed) at 0x00010000 in 2.2 seconds (effective 1249.3 kbit/s)...
Hash of data verified.

### What happened ?

If I am sending data to the USB Device, i sometimes see a flood of random traffic coming back from the TinyUSB device, even though I have no function defined to send data back and only reading.

### How to reproduce ?

1. Compile the attached the sketch
2. Upload it to an ESP32-S3 board
3. Send a lot of MIDI messages to the device - first you will see only events to the device and then it will start flooding the USB MIDI.

### Screenshots

[image]

DatanoiseTV, 2024-07-01:
It seems to be a bug in the MIDI library. When working with raw tud_midi_ API from TinyUSB, it works as expected.

hathach, 2024-08-08:
please follow the template to attached compile log as txt for readibility

closed 2025-05-21 by hathach