adafruit/Adafruit_TinyUSB_Arduino #436: USB MSC on Not Attaching Feather ESP32-S3 Reverse TFT

brentru, 2024-07-01:
### Operating System

MacOS

### Arduino IDE version

1.8.19

### Board

Adafruit Feather ESP32-S3 Reverse TFT

### ArduinoCore version

latest ESP32 BSP

### TinyUSB Library version

latest

### Sketch as ATTACHED TXT

example - msc_external_flash 

### Compiled Log as ATTACHED TXT

[compile.txt](https://github.com/user-attachments/files/16058399/compile.txt)

### What happened ?

@hathach  Hi - a user reported that their board (Adafruit Feather ESP32-S3 w/PSRAM) running WipperSnapper Arduino failed to attach USB. I'm able to reproduce this with a Feather ESP32-S3 Reverse TFT and the msc_external_flash.ino example from this library.

The device can initialize and read the flash contents but it never mounts/shows up in macOS Finder:
```
Adafruit TinyUSB Mass Storage External Flash example
JEDEC ID: 0x204016
Flash size: 960 KB
Opening root
Flash contents:
         0 .fseventsd/
         0 .metadata_never_index
         0 .Trashes
       262 wipper_boot_out.txt
       237 secrets.json
       237 ignore.json
```

For reference, the code section which initializes USB-MSC within WipperSnapper is here: https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/blob/main/src/provisioning/tinyusb/Wippersnapper_FS.cpp#L194

Related issue - https://github.com/adafruit/Adafruit_Wippersnapper_Arduino/issues/591

### How to reproduce ?

Upload the `msc_external_flash` example to an Adafruit Feather ESP32-S3 Rev. TFT.

### Debug Log

[idf-log.txt](https://github.com/user-attachments/files/16058400/idf-log.txt)

### Screenshots

n/a

brentru, 2024-07-09:
@hathach Hi - could you take a look at this issue?

hathach, 2024-07-11:
@brentru I have tested with feather s3 tft with board selection is reservetft (sinnce I don't have reserve version). **It works just fine with my Linux machine, msc mounted created new file is OK**. I notice that I have to run https://github.com/adafruit/Adafruit_SPIFlash/blob/master/examples/SdFat_format/SdFat_format.ino first since my board does not pre-formatted.

I am not sure what is missing here, @brentru can you 
- run `sudo dmesg -w` as well as lsusb then `lsusb -v -d 239a:` here, maybe macos host is picky. 
- try with my compiled uf2 to test with [msc_external_flash.ino.zip](https://github.com/user-attachments/files/16171844/msc_external_flash.ino.zip)
- attached your `msc_external_flash.ino.bin` here, I will try to test your comipled binary to see if that works on my pc.

brentru, 2024-07-22:
@hathach  Sorry for the delay, was wrapping WipperSnapper for Python. Here are my test notes..

>  notice that I have to run https://github.com/adafruit/Adafruit_SPIFlash/blob/master/examples/SdFat_format/SdFat_format.ino first since my board does not pre-formatted.

I tried this, and came to the same error.

> run sudo dmesg -w when reset board to run application

Output: https://gist.github.com/brentru/b46eccdbcd358e6321ef9fddd5c956a7

> lsusb then lsusb -v -d 239a:

Output: https://gist.github.com/brentru/33d27ea032493c45982d8a9969928acd

> try with my compiled uf2 to test with [msc_external_flash.ino.zip](https://github.com/user-attachments/files/16171844/msc_external_flash.ino.zip)

This works as-expected.

> attached your msc_external_flash.ino.bin here, I will try to test your comipled binary to see if that works on my pc.

Attached! 
[msc_external_flash.ino.bin.zip](https://github.com/user-attachments/files/16336939/msc_external_flash.ino.bin.zip)

brentru, 2024-07-23:
@hathach Noting that I'm also seeing this issue on a QT Py ESP32-S2 with firmware compiled by PlatformIO IDE

brentru, 2024-08-09:
@hathach Getting the same issue on an ESP32-S2 QT PY, may be a macOS thing?

hathach, 2024-08-27:
@brentru sorry for huge delay, I tested again with latest of everything. And indeed, there is an issue enumerating with macos (tested on my x86 macos 12.7.6). As far as I could remember, arduino-esp32 core will always start with usb pin assigned to jtag usb, but then switch to otg usb if needed. For Linux/Windows, it will renumerate after a short bus reset signal, however for macos, it looks like the reset interval is too short ~ 20us, which is indeed not long enough as reset accordingly to USB spec. I am sure it previoulsy work with macos, maybe core change something in v3. Give me a bit of time, I am working on a PR (either to our lib or esp32 core) for this.

PS: meanwhile, can you verify that when unplugging the very same board to other non-macos machine, it could show up, maybe trying an android ?

[image]

brentru, 2024-08-27:
> PS: meanwhile, can you verify that when unplugging the very same board to other non-macos machine, it could show up, maybe trying an android ?

Same board works OK on Windows 11

hathach, 2024-08-27:
> > PS: meanwhile, can you verify that when unplugging the very same board to other non-macos machine, it could show up, maybe trying an android ?
> 
> Same board works OK on Windows 11

Thanks for testing, I think we found the root cause, will work on the fix real soon.

brentru, 2024-08-29:
@hathach  Okay, keep me in the loop as I am still seeing reports of this from people running WipperSnapper.

hathach, 2024-08-30:
This help to prevent S3 to be enumerated as USB JTAG while waitinf for 2nd reset (200-500 ms) https://github.com/adafruit/tinyuf2/pull/406 . Now it enumerate as arduino, however it is too fast with macOS that device is fully enumerated before `setup()/msc.begin()` is called --> mising interface. We will need to forcing re-enumeration, I am working on that for another PR.

brentru, 2024-09-03:
Okay, thanks for the update!

hathach, 2024-09-05:
@brentru this should be fixed by #454 which update tinyusb to be able to force re-enumeration with esp32s3. I will make an PR to arduino-esp32 to bump up tinyuf2 to prevent jtag enumeration, but it is optional fix :). Please try again to see if that works for you. In WipperSnapper library, you may also need to force enumeration as well since S3 is very fast, by the time we call usb_msc.begin(), the enumeration is already complete (without msc driver)

closed 2024-09-05 by hathach

brentru, 2024-09-05:
> @brentru this should be fixed by https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/454 which update tinyusb to be able to force re-enumeration with esp32s3. I will make an PR to arduino-esp32 to bump up tinyuf2 to prevent jtag enumeration, but it is optional fix :)

I'll try with the TinyUSB update only, first, then the arduino-esp32 fix

> you may also need to force enumeration as well since S3 is very fast, by the time we call usb_msc.begin(), the enumeration is already complete (without msc driver)

Do you have an example of doing this?