adafruit/Adafruit_TinyUSB_Arduino #440: TinyUSB - RP2040 Compilation Error

Jordessey, 2024-07-17:
### Operating System

Windows 11

### Arduino IDE version

2.3.2

### Board

Raspberry Pi RP2040

### ArduinoCore version

4.1.5

### TinyUSB Library version

3.3.1

### Sketch as ATTACHED TXT

examples/HID/hid_gamepad.ino

### Compiled Log as ATTACHED TXT

[TinyUSB Compliation Error.txt](https://github.com/user-attachments/files/16262458/TinyUSB.Compliation.Error.txt)

### What happened ?

Unable to compile HID Gamepad example on MBED rp RP2040 

### How to reproduce ?

will not compile, attempted reinstalling Arduino IDE, reinstalling library and board core

touchgadget, 2024-07-23:
> ArduinoCore version: 4.1.5

Support for mbed_rp2040 is listed as experimental. I suggest using https://github.com/earlephilhower/arduino-pico.

KerchumA222, 2024-08-09:
I have the same compilation errors when building a PlatformIO project with this library.