adafruit/Adafruit_TinyUSB_Arduino #442: Possible bug with Waveshare RP2040 PiZero and TinyUSB / PIO-USB

sonik-br, 2024-07-29:
### Operating System

Windows 11

### Arduino IDE version

Arduino 1.8.19

### Board

Waveshare RP2040 PiZero

### ArduinoCore version

Raspberry Pi Pico/RP2040 (earlephilhower) 3.9.4

### TinyUSB Library version

3.2.2

### Sketch as ATTACHED TXT

examples/DualRole/Simple/device_info

Only changed the PIN_USB_HOST_DP to 6, and commented out the PIN_5V stuff.
Added delay to setup() and setup1() to have some time to power the board and open the serial monitor.

[device_info.txt](https://github.com/user-attachments/files/16413928/device_info.txt)
[usbh_helper.txt](https://github.com/user-attachments/files/16413929/usbh_helper.txt)

### Compiled Log as ATTACHED TXT

[compile_log.txt](https://github.com/user-attachments/files/16413991/compile_log.txt)

### What happened ?

Some USB devices are not recognized when connected after powering the pico. Usb disconnect/reconnect also does not work.
If a device is connected prior to powering the pico, then powering the pico, it gets detected as expected.

Same code works fine on other common Picos.
Only seem to happen on the Waveshare RP2040 PiZero.
Tested on it's on-board usb-c pio port and also bydirectly connecting a female usb connector to it's GPIO pins (6 and 7).

Tested devices that are **working**:
* Sony DualSense controller
* Microsoft One Wired contoller

Tested devices that are **not working**:
* Sony DualShock 3 controller
* Sony DualShock 4 controller
* Microsoft Xbox 360 Wired contoller
* Nintendo Switch Pro contoller

Using PIO-USB version 0.5.3

### How to reproduce ?

Need to run the sketch and connect some of the problematic usb devices.

hathach, 2024-08-08:
we don't have this specific hardware to test/troubleshoot. If you figure out the root cause, feel free to make an PR

closed 2024-08-08 by hathach

sonik-br, 2024-08-08:
@hathach Perhaps related to the way the pins are wired and the resistors?
Using USB-PIO on any other set of GPIO pins does work. Problem only happens if using the board's designated pio-usb port.

[image]

I'm a layman in electronics but you might spot a problem here (if there's one).

domiluci, 2024-09-23:
> @hathach Perhaps related to the way the pins are wired and the resistors? Using USB-PIO on any other set of GPIO pins does work. Problem only happens if using the board's designated pio-usb port. [...]

I've been having trouble with the USB-PIO port of this [Waveshare RP2040 Pi Zero](https://www.amazon.com/dp/B0CJ6VM6H6?ref=ppx_yo2ov_dt_b_fed_asin_title) board as well. I've been trying to attach [this USB trackpad](https://www.amazon.com/gp/product/B0C2D5CG93/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) to the USB-PIO port to use with [HID Remapper](https://github.com/jfedor2/hid-remapper) by @jfedor2. The USB trackpad works perfectly fine when plugged into any normal USB port of my computer, so I know it's not the trackpad. But it won't even power up when plugged into the RP2040 Pi Zero's USB-PIO port, which is odd because it doesn't draw much current, and it definitely draws less current than the controllers and devices you've listed as working with the USB-PIO port of the RP2040 Pi Zero. So I should at least be getting power.

But, looking at this schematic, I noticed the DATA+ & DATA- lines of the USB-PIO port are being pulled up using a +3V3 supply. That makes me wonder if either the +3V3 (instead of +5V) pullups are interfering with USB device enumeration, and/or the USB-PIO port is being powered by an incompatible or underpowered source on the PCB. If not, maybe I missed something or I just got a lemon.

At any rate, I'm going to try some other devices to see if they work.