adafruit/Adafruit_TinyUSB_Arduino #447: Switch between USB Host and USB Device modes

doctea, 2024-08-04:
**Is your feature request related to a problem? Please describe.**
I am working on an RP2040 project where sometimes it would be useful to use it as a USB MIDI host, other times as a USB MIDI device.  The hardware I'm building on doesn't provide for adding the "second" USB port on GPIO, so it'd be great to be able to switch between the two without requiring recompilation.  

**Describe the solution you'd like**
Most ideal would be if it were possible to switch between host and device mode on the fly, without rebooting the MCU.  

Alternatively, being able to choose which mode to start in without requiring recompilation could work (eg a 'reboot into device mode', 'reboot into host mode' option that sets a flag in flash that chooses which mode the USB starts up in.

I appreciate this might not be a trivial feature, but it would be really useful.

samspencer5991, 2025-04-24:
This would be a very useful inclusion

hathach, 2025-05-21:
currrently we don't have plan or time to do this

closed 2025-05-21 by hathach