adafruit/Adafruit_TinyUSB_Arduino #465: Linker error on ESP32 S3 with 3.07 Arduino Core

pschatzmann, 2024-10-23:
### Operating System

Linux

### Arduino IDE version

2.3.3

### Board

ESP32S3 DEV Module

### ArduinoCore version

3.07

### TinyUSB Library version

3.3.4

### Sketch as ATTACHED TXT

Any example: e.g. your midi_test 

### Compiled Log as ATTACHED TXT

[compile_log.txt](https://github.com/user-attachments/files/17490472/compile_log.txt)

### What happened ?

Linker is failing because of multiple definitions

### How to reproduce ?

- Delete the installed library and  delete the esp32 installation to force a new installation
- Install the ESP32 core
- Install the latest version of Adafruit_TinyUSB_Arduino
- Compile any of the example sketches

### Debug Log

n/a

pschatzmann, 2024-10-24:
Cool, works like a charm now...
Thanks a lot!

wamk9, 2024-11-04:
Same error here with 3.4.0

PR #466 (2024-10-24): update tinyusb to fix esp32 build with v3.0.7
update tinyusb to commit 8b1e40c3e2447de5b4a86bbcdcc0344946a4793d which use new style of tusb_init()/tuh_init()/tud_init(). This should fix #465 building with latest esp32 core v3.0.7