adafruit/Adafruit_TinyUSB_Arduino #469: TinyUSB Video not working on ESP32 S3

pschatzmann, 2024-11-04:
### Operating System

Linux

### Arduino IDE version

2.3.3

### Board

ESP32 S3 Dev Board

### ArduinoCore version

3.07

### TinyUSB Library version

3.4.0

### Sketch as ATTACHED TXT

Pease use your example video_capture Example Sketch

### Compiled Log as ATTACHED TXT

[compile_log.txt](https://github.com/user-attachments/files/17620765/compile_log.txt)

### What happened ?

- I needed to reboot to get the right Video USB descriptor.
- Trying out the device with the VLC Media Player displays the device, but opening it gives an error 
- After unplugging and replugging the ESP32S3 the video dscriptor has diappeard

I tried the same with an RP2040 and there it is working w/o any problems.

### How to reproduce ?

See above

### Debug Log

Please provide instructions how to get to this log.
I am not getty any output via CDC

maxgerhardt, 2024-11-19:
RP2040 + Linux + `video_capture.ino` works.
[image]

```
[   77.928096] usb 1-2: new full-speed USB device number 3 using xhci_hcd
[   78.414378] usb 1-2: New USB device found, idVendor=2e8a, idProduct=000a, bcdDevice= 1.00
[   78.414388] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   78.414392] usb 1-2: Product: Pico
[   78.414395] usb 1-2: Manufacturer: Raspberry Pi
[   78.414398] usb 1-2: SerialNumber: E66038B7133C5D30
[   78.440293] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[   78.440459] usbcore: registered new interface driver cdc_acm
[   78.440466] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   78.442097] mc: Linux media interface: v0.10
[   78.463797] videodev: Linux video capture interface: v2.00
[   78.481348] usb 1-2: Found UVC 1.50 device Pico (2e8a:000a)
[   78.493684] usbcore: registered new interface driver uvcvideo
```

maxgerhardt, 2024-11-19:
I'm looking into the USB packet flow when using Windows, and Windows issues a special `URB_FUNCTION_GET_MS_FEATURE_DESCRIPTOR` packets, which seems to be linked to [Microsoft OS Feature Descriptors](https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/usb/ns-usb-_urb_os_feature_descriptor_request). Maybe TinyUSB doesn't return something here that Microsoft Windows definitely needs or wants to know, and this the RP2040 example fails on Windows but not on Linux.

maxgerhardt, 2024-11-19:
There's also a discussion open about it: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/discussions/417

umeiko, 2024-11-20:
I'm looking forword this fix, it will be much great if rp2040 can have full uvc support