adafruit/Adafruit_TinyUSB_Arduino #471: When I plug the wired Logitech G903 into the USB host of RP2040, it crashes

FBMNBAWA, 2024-11-09:
### Operating System

Windows 10

### Arduino IDE version

2.33

### Board

Raspberry pico

### ArduinoCore version

4.11

### TinyUSB Library version

3.40

### Sketch as ATTACHED TXT

examples/DualRole/HID
/hid_device_report

### Compiled Log as ATTACHED TXT

Pico and mouse are both stuck, and can only be restored by restarting Pico. After a while, they are stuck again

### What happened ?

Pico and mouse are both stuck, and can only be restored by restarting Pico. After a while, they are stuck again. But if you use a receiver connection, there won't be any problem

### How to reproduce ?

Connect a wired connection to Logitech G903 on the USB host of RP2040

hathach, 2024-11-20:
can you try to change CFG_TUSB_DEBUG = 2, then get the LOG via Serial1 (hw uart) and post it here.

FBMNBAWA, 2024-11-20:
Thank you for answering my question. It has been bothering me for a long time

FBMNBAWA, 2024-11-20:
on EP 00 with 0 bytes: OK

[1] Claimed EP 0x81
  Queue EP 81 with 8 bytes ... 
OK
HID set config: itf = 1
HID Set Idle 
[1:1] Class Request: 21 0A 00 00 01 00 00 00 
on EP 81 with 0 bytes: FAILED
HID xfer callback
  Get Report callback (1, 0)
[1] Claimed EP 0x81
  Queue EP 81 with 8 bytes ... 
OK
on EP 00 with 8 bytes: FAILED
[1:1] Control FAILED, xferred_bytes = 8
21 0A 00 00 01 00 00 00 

HID Get Report Descriptor
[1:1] Get Descriptor: 81 06 00 22 01 00 97 00 
on EP 81 with 0 bytes: FAILED
HID xfer callback
  Get Report callback (1, 0)
[1] Claimed EP 0x81
  Queue EP 81 with 8 bytes ... 
OK
on EP 00 with 8 bytes: FAILED
[1:1] Control FAILED, xferred_bytes = 8
81 06 00 22 01 00 97 00

blaxkwidow, 2025-02-01:
same issue here, did you manage to fix the problem ? @FBMNBAWA

FBMNBAWA, 2025-02-01:
no @blaxkwidow

blaxkwidow, 2025-02-01:
it cant be a mouse issue since I'm using a different brand ( razer ) 
Im trying to make the rp2040 with host a "passthrough" device ( for mice ) however the board and the mouse gets off after few seconds ( randomly ) and it will not function or i cant even upload a new code until its fully restarted ( reconnected to the pc )

experiencing the mouse issue on my own code currently:
```
#include "usbh_helper.h"

uint8_t const desc_hid_report[] = {
    TUD_HID_REPORT_DESC_MOUSE(HID_REPORT_ID(1))
};

Adafruit_USBD_HID usb_hid;

void setup() {
    Serial.begin(115200);
    
    if (!TinyUSBDevice.isInitialized()) {
        TinyUSBDevice.begin(0);
    }

    usb_hid.setPollInterval(0);
    usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
    usb_hid.begin();

    rp2040_configure_pio_usb();
    USBHost.begin(1);

    while (!TinyUSBDevice.mounted()) delay(1);
}

void loop() {
    USBHost.task();
}

extern "C" {
    void tuh_hid_mount_cb(uint8_t dev_addr, uint8_t instance, uint8_t const* desc_report, uint16_t desc_len) {
        uint8_t const itf_protocol = tuh_hid_interface_protocol(dev_addr, instance);
        if (itf_protocol == HID_ITF_PROTOCOL_MOUSE) {
            if (!tuh_hid_receive_report(dev_addr, instance)) {
                Serial.printf("Failed to start receiving report\r\n");
            }
        }
    }

    void tuh_hid_report_received_cb(uint8_t dev_addr, uint8_t instance, uint8_t const* report, uint16_t len) {
        if (len >= 3) {
            uint8_t buttons = report[0];
            int8_t x = report[4];
            int8_t y = report[6];
            
            if (usb_hid.ready()) {
                usb_hid.mouseReport(1, buttons, x, y, report[3], 0);
            }
        }

        if (!tuh_hid_receive_report(dev_addr, instance)) {
            Serial.printf("Failed to request next report\r\n");
        }
    }

    void tuh_hid_umount_cb(uint8_t dev_addr, uint8_t instance) {
        Serial.printf("Device unmounted: %d\r\n", dev_addr);
    }
}
```

@FBMNBAWA my discord is hafef if you would like to help each other fix this issue

FBMNBAWA, 2025-02-02:
Yes, I encountered the same problem, which is exactly the same as your description @blaxkwidow

hathach, 2025-05-22:
please try again with v3.5.0 and pio 0.7.2, and upload your compiled log as txt file

hathach, 2025-07-17:
closed due to lack of response. probably the issue with pio-usb and tinyusb host driver, there are quite a bit of update. Please try again, and re-open a new issue if it still exists