adafruit/Adafruit_TinyUSB_Arduino #473: Linker Error on esp32s3 with platformIO

umeiko, 2024-11-19:
### Operating System

Windows 11

### Arduino IDE version

platformIO

### Board

esp32s3 dev module

### ArduinoCore version

espressif32 @ 6.6.0

### TinyUSB Library version

3.4.0

### Sketch as ATTACHED TXT

The examlpe `video_capture.ino`

### Compiled Log as ATTACHED TXT

[Processing esp32-s3-fh4r2 (platform.txt](https://github.com/user-attachments/files/17809265/Processing.esp32-s3-fh4r2.platform.txt)

### What happened ?

Just can't compile, multiple definition warnings ane ld returns error.

### How to reproduce ?

1. create a new project with platformIO, choose esp32s3 board.
2. add `adafruit/Adafruit TinyUSB Library` to this project
3. copy `video_capture.ino` example into `main.cpp`
4. compile.

umeiko, 2024-11-19:
there also some chat about that `https://community.platformio.org/t/new-tinyusb-doesnt-allow-compile/44281`

hathach, 2024-11-20:
there is no arduino-esp32 6.6.0, please specify the correct arduino core verion

umeiko, 2024-11-20:
> there is no arduino-esp32 6.6.0, please specify the correct arduino core verion

seems this problem shown in platformIO, which not have a arduino core verion but only `esp32 arduino` version

hathach, 2024-11-27:
closed since arduino-esp32 version is not provided and issue is not reproducible with latest core version of 3.0.7

closed 2024-11-27 by hathach

dhebbeker, 2024-12-23:
I have experienced the same issue. Note that I (and probably the OP) have been using Espressif's Arduino-ESP32 as core.

Here is the generated dependency tree:

```plainplatformio pkg list --environment esp32-s3-devkitc-1 

Resolving esp32-s3-devkitc-1 dependencies...
Platform espressif32 @ 6.9.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20017.0 (required: platformio/framework-arduinoespressif32 @ ~3.20017.0)
├── tool-esptoolpy @ 1.40501.0 (required: platformio/tool-esptoolpy @ ~1.40501.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5)

Libraries
└── Adafruit TinyUSB Library @ 3.4.0 (required: adafruit/Adafruit TinyUSB Library @ 3.4.0)
│   ├── Adafruit SPIFlash @ 5.0.1 (required: Adafruit SPIFlash)
│   │   ├── Adafruit NeoPixel @ 1.12.3 (required: Adafruit NeoPixel)
│   │   └── SdFat - Adafruit Fork @ 2.2.3 (required: SdFat - Adafruit Fork)
│   ├── MIDI Library @ 5.0.2 (required: MIDI Library)
```

This gives the following errors:

```plain
Linking .pio/build/esp32-s3-devkitc-1/firmware.elf
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tusb_inited':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:62: multiple definition of `tusb_inited'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libdeps/esp32-s3-devkitc-1/Adafruit TinyUSB Library/src/tusb.c:102: first defined here
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tu_desc_find':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:81: multiple definition of `tu_desc_find'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libdeps/esp32-s3-devkitc-1/Adafruit TinyUSB Library/src/tusb.c:137: first defined here
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tu_desc_find2':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:91: multiple definition of `tu_desc_find2'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libdeps/esp32-s3-devkitc-1/Adafruit TinyUSB Library/src/tusb.c:145: first defined here
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tu_desc_find3':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:101: multiple definition of `tu_desc_find3'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libdeps/esp32-s3-devkitc-1/Adafruit TinyUSB Library/src/tusb.c:153: first defined here
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tu_edpt_claim':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:115: multiple definition of `tu_edpt_claim'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libdeps/esp32-s3-devkitc-1/Adafruit TinyUSB Library/src/tusb.c:165: first defined here
/home/dhebbeker/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /home/dhebbeker/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/lib/libarduino_tinyusb.a(tusb.c.obj): in function `tu_edpt_release':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino_tinyusb/tinyusb/src/tusb.c:133: multiple definition of `tu_edpt_release'; .pio/build/esp32-s3-devkitc-1/lib663/libAdafruit TinyUSB Library.a(tusb.c.o):/home/dhebbeker/pro/time-tracker/.pio/libd [... cut ...]
  "description": "Arduino Wiring-based Framework for the Espressif ESP32, ESP32-S and ESP32-C series of SoCs",
  "keywords": [
    "framework",
    "arduino",
    "espressif",
    "esp32"
  ],
  "license": "LGPL-2.1-or-later",
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/arduino-esp32"
  }
}
```

When using  v.3.3.4 instead of v.3.4.0 allows to compile & link. Haven't tested functionality.

``` diff
lib_deps = 
-    adafruit/Adafruit TinyUSB Library@3.4.0
+    adafruit/Adafruit TinyUSB Library@3.3.4
```