adafruit/Adafruit_TinyUSB_Arduino #474: error: static assertion failed: CFG_TUSB_RHPORT0_MODE/CFG_TUSB_RHPORT1_MODE must be defined on RP2040 since specific update

doctea, 2024-11-19:
### Operating System

Windows 11

### Arduino IDE version

VSCode+Platformio

### Board

RP2040

### ArduinoCore version

n/a

### TinyUSB Library version

f13b57a01aaed4fe51b3000cb3ce89ad8c8c83dc

### Sketch as ATTACHED TXT

n/a

### Compiled Log as ATTACHED TXT

```
In file included from C:\Users\doctea\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp:37:
C:\Users\doctea\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp: In function 'void __usb(void*)':
.pio\libdeps\pcb\Adafruit TinyUSB Library\src/tusb.h:146:53: error: static assertion failed: CFG_TUSB_RHPORT0_MODE/CFG_TUSB_RHPORT1_MODE must be defined
  146 |   #define _tusb_init_arg0()        TU_VERIFY_STATIC(false, "CFG_TUSB_RHPORT0_MODE/CFG_TUSB_RHPORT1_MODE must be defined")
      |                                                     ^~~~~
.pio\libdeps\pcb\Adafruit TinyUSB Library\src/common/tusb_compiler.h:39:31: note: in expansion of macro '_tusb_init_arg0'
   39 | #define TU_STRCAT(a, b)       a##b                ///< concat without expand
      |                               ^
.pio\libdeps\pcb\Adafruit TinyUSB Library\src/common/tusb_compiler.h:42:31: note: in expansion of macro 'TU_STRCAT'
   42 | #define TU_XSTRCAT(a, b)      TU_STRCAT(a, b)     ///< expand then concat
      |                               ^~~~~~~~~
.pio\libdeps\pcb\Adafruit TinyUSB Library\src/common/tusb_compiler.h:127:43: note: in expansion of macro 'TU_XSTRCAT'
  127 | #define TU_FUNC_OPTIONAL_ARG(func, ...)   TU_XSTRCAT(func##_arg, TU_ARGS_NUM(__VA_ARGS__))(__VA_ARGS__)
      |                                           ^~~~~~~~~~
.pio\libdeps\pcb\Adafruit TinyUSB Library\src/tusb.h:151:46: note: in expansion of macro 'TU_FUNC_OPTIONAL_ARG'
  151 | #define tusb_init(...)                       TU_FUNC_OPTIONAL_ARG(_tusb_init, __VA_ARGS__)
      |                                              ^~~~~~~~~~~~~~~~~~~~
C:\Users\doctea\.platformio\packages\framework-arduinopico\libraries\FreeRTOS\src\variantHooks.cpp:468:5: note: in expansion of macro 'tusb_init'
  468 |     tusb_init();
      |     ^~~~~~~~~
```

### What happened ?

Updated to newer version of TinyUSB on my RP2040 project, compile fails with the output below.  I've done a `git bisect` and found that `f13b57a01aaed4fe51b3000cb3ce89ad8c8c83dc is the first bad commit`.  Hope this helps someone!

### How to reproduce ?

Compile on RP2040 with any version of TinyUSB after commit f13b57a01aaed4fe51b3000cb3ce89ad8c8c83dc.

doctea, 2024-11-20:
> what is the arduino core version ?

I'll check precise version and get back to you.  I believe it should be the latest version of the earlephilhower arduino-pico core, though.

doctea, 2024-11-21:
My platformio.ini says `platform = https://github.com/maxgerhardt/platform-raspberrypi` and it was installed on the 20th November -- is that all you need to know?

mztulip, 2025-01-28:
I have the same problem. Solutions for me is to add definition which is requested. My code with usb device works. I hope that this solution helps, because I spend a lot of time trying to understand what happens and solve it.
Here is my platformio.ini with added CFG_TUSB_RHPORT0_MODE define:
```
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
build_flags = -DUSE_TINYUSB -DCFG_TUSB_RHPORT0_MODE
lib_deps=https://github.com/pschatzmann/Adafruit_TinyUSB_Arduino.git#Audio

[env:pico]
board = pico

```

hathach, 2025-02-06:
> My platformio.ini says `platform = https://github.com/maxgerhardt/platform-raspberrypi` and `board_build.core = earlephilhower` and it was installed on the 20th November -- is that all you need to know?

please specify the core release version of arduino-pico https://github.com/earlephilhower/arduino-pico/releases

mztulip, 2025-02-09:
My core version is 4.4.1

hathach, 2025-02-10:
1.4 is too old, please update core version to latest and re-test. If you couldn't update the core, try to reduce the tinyusb lib to version that work with the old core.

closed 2025-05-22 by hathach