adafruit/Adafruit_TinyUSB_Arduino #482: clearConfiguration() prevents mounting

JeffCalwood, 2024-12-22:
### Operating System

Windows 10

### Arduino IDE version

2.3.4

### Board

XIAO SAMD21

### ArduinoCore version

Seeed SAMD 1.8.5 (old but latest I think)

### TinyUSB Library version

3.4.2

### Sketch as ATTACHED TXT

[modified hid_boot_mouse sketch.txt](https://github.com/user-attachments/files/18222023/modified.hid_boot_mouse.sketch.txt)

### Compiled Log as ATTACHED TXT

see attached, recorded with 
[hid_boot_mouse_with_cdc_disabled.txt](https://github.com/user-attachments/files/18222057/hid_boot_mouse_with_cdc_disabled.txt)
#define CFG_TUSB_DEBUG 3

### What happened ?

This is based on the hid_boot_mouse.ini example sketch in library version 3.4.2 . Adding a
```
TinyUSBDevice.clearConfiguration();
```
before the usb.begin() will stop the device from mounting. Am I doing this wrong?
(I removed all serial logging and substituted with BUILDIN_LED feedback. See attached sketch)

### How to reproduce ?

* run sketch, comment or un-comment line 66 ( clearConfiguration() call )
* blinking LED means USB mounted

### Debug Log

n/a

### Screenshots

n/a

JeffCalwood, 2024-12-22:
I believe the root cause for my findings above is a conflict between the TinyUSB library that came 
* pre-installed with XIAO SAMD21, and
* library v3.4.2 manually installed by me.
TL;DR: Cleaning up that conflict fixed this bug (and it could/should be closed). BUT I wonder whether this was just me or if others could run into it. The manifestation was VERY subtle.

Since I wasn't given an official option to upgrade the XIAO core, I ended up moving the manually installed (latest) version into the core folder (it uses a different folder name, though). That fixed a few subtle OTHER issues I was also encountering. To wrap up today's troubleshooting, I re-tested the above sketch and found that it works after the library "clean up". I will leave it up to the developers whether you just want to close this bug. If I can help with additional info/details I will try.

hathach, 2024-12-25:
clearConfiguration() clear all configuration to allow user get rid of serial interface (some need that). It still enumerate but without any class driver, function works as expected.

closed 2024-12-25 by hathach