adafruit/Adafruit_TinyUSB_Arduino #487: 3.4.0 introduces an issue on my ESP32-S3 that causes the USB device to not enumerate correctly

i-am-shodan, 2025-01-14:
### Operating System

Windows 11

### Arduino IDE version

PlatformIO

### Board

LilyGo Dongle S3

### ArduinoCore version

https://github.com/platformio/platform-espressif32.git

### TinyUSB Library version

3.4.0+

### Sketch as ATTACHED TXT

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>

void setup() {
  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  Serial.begin(115200);
}

void loop() {
  Serial.println("Loop!");
  Serial.flush();
  delay(1000);
}
```

```
[env:esp-32-s3]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = arduino
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
build_unflags = 
	-DARDUINO_USB_MODE=1
build_flags = 
	-std=gnu++2a
	-DARDUINO_USB_MODE=0
	-DARDUINO_USB_CDC_ON_BOOT=1
        -Wl,-z,muldefs
	-D CFG_TUD_ENABLED
lib_deps = 	https://github.com/i-am-shodan/Adafruit_TinyUSB_Arduino
```

### Compiled Log as ATTACHED TXT

[attached.txt](https://github.com/user-attachments/files/18408980/attached.txt)

### What happened ?

Starting in 3.4.0 the sketch above no longer results in a USB device appearing/COM port in Windows 11.

* 3.1.5 works
* 3.3.0 works
* 3.3.3 works
* 3.3.4 works
* 3.4.0 does NOT work
* 3.4.1 does NOT work

### How to reproduce ?

Compile, flash and plug in

hathach, 2025-01-17:
please specify your exact arduino-esp32 version. If you don't know, try to compile with Arduino IDE instead.

i-am-shodan, 2025-01-17:
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = 
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.5
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1

hathach, 2025-05-21:
not reproducible with arduino IDE + arduino-esp32 v3.20 + tinyusb 3.5.0. Please update your set up and try again.

closed 2025-05-21 by hathach