adafruit/Adafruit_TinyUSB_Arduino #500: Test Builds are failing, including wrong SdFat library.

SmittyHalibut, 2025-03-01:
### Operating System

Linux

### Arduino IDE version

PlatformIO, but irrelevant. This is in the github build check system.

### Board

n/a

### ArduinoCore version

feather_rp2040_tinyusb

### TinyUSB Library version

main branch

### Sketch as ATTACHED TXT

n/a, but this is happening while processing a pull request: #493 

### Compiled Log as ATTACHED TXT

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/actions/runs/13607387549/job/38040486407?pr=493#logs

Relevant bit:
```
msc_external_flash.ino rp2040:rp2040:adafruit_feather:flash=8388608_0,freq=120,dbgport=Disabled,dbglvl=None,usbstack=tinyusb built output
  Used library             Version Path
  SPI                      1.0     /home/runner/.arduino15/packages/rp2040/hardware/rp2040/4.4.4/libraries/SPI
  SdFat                    2.3.0   /home/runner/.arduino15/packages/rp2040/hardware/rp2040/4.4.4/libraries/SdFat
  Adafruit SPIFlash        5.0.1   /home/runner/Arduino/libraries/Adafruit_SPIFlash
  Adafruit TinyUSB Library 3.4.2   /home/runner/work/Adafruit_TinyUSB_Arduino/Adafruit_TinyUSB_Arduino
  
  Used platform Version Path
  rp2040:rp2040 4.4.4   /home/runner/.arduino15/packages/rp2040/hardware/rp2040/4.4.4
  In file included from /home/runner/work/Adafruit_TinyUSB_Arduino/Adafruit_TinyUSB_Arduino/examples/MassStorage/msc_external_flash/msc_external_flash.ino:27:
  /home/runner/Arduino/libraries/Adafruit_SPIFlash/src/Adafruit_SPIFlash.h:37:2: error: #error USE_BLOCK_DEVICE_INTERFACE must be defined to 1 in SdFatConfig.h. Make sure you use the Adafruit Fork at 'https://github.com/adafruit/SdFat'
     37 | #error USE_BLOCK_DEVICE_INTERFACE must be defined to 1 in SdFatConfig.h. Make sure you use the Adafruit Fork at 'https://github.com/adafruit/SdFat'
        |  ^~~~~
  /home/runner/Arduino/libraries/Adafruit_SPIFlash/src/Adafruit_SPIFlash.h:54:2: error: #error FAT12_SUPPORT must be set to 1 in SdFat SdFatConfig.h. Make sure you use the Adafruit Fork at 'https://github.com/adafruit/SdFat'
     54 | #error FAT12_SUPPORT must be set to 1 in SdFat SdFatConfig.h. Make sure you use the Adafruit Fork at 'https://github.com/adafruit/SdFat'
        |  ^~~~~
  Error during build: exit status 1
	msc_external_flash_sdcard.ino ❌
```

### What happened ?

The Build Checks are building against the wrong SdFat library.

In "Used library", note that it's referencing the library in `.arduino16/packages` not in `Arduino/libraries`

Then in the build, it says it's built against the wrong SdFat.

### How to reproduce ?

Submit a pull-request against `adafruit:master`.  The build checks fail this way.

SmittyHalibut, 2025-03-01:
As a work-around for this to get #493 to pass builds, I excluded the failing tests by adding `feather_rp2040_tinyusb` to `.skip.txt`.  If #493 has been merged when you're looking into this, you'll need to remove those skips to reproduce this.

See commit 5311fb83c4407041538d91791714457d66a81ae4

hathach, 2025-05-21:
this should be fixed now, thanks for reproting the issue