adafruit/Adafruit_TinyUSB_Arduino #506: Platformio and `tud_vendor_control_xfer_cb`

snaens, 2025-03-10:
### Operating System

Linux

### Arduino IDE version

Platformio

### Board

pi pico

### ArduinoCore version

earlephilower

### TinyUSB Library version

latest

### Sketch as ATTACHED TXT

```ini
[env:pico]

platform = raspberrypi
board = rpipico
framework = arduino
board_build.core = earlephilhower

lib_archive = false

build_flags =
  -DUSE_TINYUSB
```

```
.
├── platformio.ini
└── src
    ├── Adafruit_USBD_I2C.cpp
    ├── Adafruit_USBD_I2C.h
    └── i2c_tiny_usb_adapter.ino
```

### Compiled Log as ATTACHED TXT

```
<truncated>
PLATFORM: Raspberry Pi RP2040 (1.15.0+sha.e95c19c) > Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
Converting i2c_tiny_usb_adapter.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 69 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit TinyUSB Library @ 3.4.1
|-- Wire @ 1.0
Compiling .pio/build/pico/src/i2c_tiny_usb_adapter.ino.cpp.o
```

### What happened ?

The way `tud_vendor_control_xfer_cb` is linked in [Adafruit_USBD_WebUSB.cpp](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/arduino/webusb/Adafruit_USBD_WebUSB.cpp#L254-L255) causes runtime-only errors that are remarkably hard to track down, as they cause no compilation hiccups.

Emptying `src/arduino/webusb/Adafruit_USBD_WebUSB.*` is a workaround.

[This platformio thread](https://community.platformio.org/t/linking-libraries-with-weak-symbols/6302/7) recomended trying `lib_archive = false`, which didn't work for me
or using `extern "C" {...}` around the `tud_vendor_control_xfer_cb` function, which also didn't work, I think because the function is a class member?

### How to reproduce ?

Try to compile the [i2c example](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/Vendor/i2c_tiny_usb_adapter) in platformio.

hathach, 2025-05-21:
does issue occur when compiling with Arduino IDE ?

snaens, 2025-05-22:
No, only in platformio.

This function seems to be at fault.

https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/59d4393012c4972e7904adc181c2747b44035085/src/arduino/webusb/Adafruit_USBD_WebUSB.cpp#L254-L256

Perhaps platformio doesn't define one of the compiler variables used in this file?
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/59d4393012c4972e7904adc181c2747b44035085/src/common/tusb_compiler.h#L133-L138

hathach, 2025-05-22:
then you could compare the compile & linking log of Arduino IDE with the one used by platformio, may be it misses a flag or two.

snaens, 2025-05-22:
Hm, strange, I tried to reproduce the problem,
but it seems to no longer be an issue.

Perhaps platformio fixed something?

Thanks for your efforts!

closed 2025-05-22 by snaens