adafruit/Adafruit_TinyUSB_Arduino #507: MAX3421E examples won't compile for esp32

BlitzCityDIY, 2025-03-11:
### Operating System

Windows 11

### Arduino IDE version

1.8.19

### Board

Adafruit QT Py ESP32

### ArduinoCore version

esp32 3.2.0-RC1

### TinyUSB Library version

3.4.3

### Sketch as ATTACHED TXT

[Arduino_USB_Host_BFF_DeviceInfo.txt](https://github.com/user-attachments/files/19198319/Arduino_USB_Host_BFF_DeviceInfo.txt)

### Compiled Log as ATTACHED TXT

[output.txt](https://github.com/user-attachments/files/19198354/output.txt)

### What happened ?

i came across this issue after an Arduino CI run on the Learn repo: https://github.com/adafruit/Adafruit_Learning_System_Guides/actions/runs/13796988406/job/38591397494#step:7:841

the error is ```
error: #error "MCUs are only supported with CFG_TUH_MAX3421 enabled"
    395 |   #error "MCUs are only supported with CFG_TUH_MAX3421 enabled"
```

both examples that failed test against esp32 only.

locally, i recreated the issue. i tried including #define CFG_TUH_MAX3421 1 in the sketch but it still throws the same error. also tried compiling for the qtpy esp32-s3 but got the same error.

when i compile for the feather m4, it does not throw the error so it seems to be espressif only.

### How to reproduce ?

the same issue also occurs with the example from the library that the learn sketch is based on (dualrole - simple - device_info_max3421e)

1. open example (dualrole - simple - device_info_max3421e)
2. select adafruit qtpy esp32
3. compile
4. #error "MCUs are only supported with CFG_TUH_MAX3421 enabled occurs

hathach, 2025-03-20:
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/pull/510 may help to fix this. It seems to compile fine with currennt arduino-esp32, but seems to have missing atomic lib issue with gcc 14. Which i will try to fix later.

BlitzCityDIY, 2025-03-20:
thanks @hathach ! just tested locally and the examples compile now

hathach, 2025-03-24:
no problems, I am glad it works for you

closed 2025-03-25 by tyeth