adafruit/Adafruit_TinyUSB_Arduino #514: device_info example on Adafruit Feather RP2040 fails on enumeration attempt

John-Leitch, 2025-04-09:
### Operating System

Windows 11

### Arduino IDE version

2.3.5

### Board

Adafruit Feather RP2040 with USB Type A Host

### ArduinoCore version

Raspberry Pi Pico/RP2040/RP2350 4.5.1

### TinyUSB Library version

3.4.4

### Sketch as ATTACHED TXT

[device_info.ino.txt](https://github.com/user-attachments/files/19660274/device_info.ino.txt)

### Compiled Log as ATTACHED TXT

[compile.log](https://github.com/user-attachments/files/19660267/compile.log)

### What happened ?

I'm encountering an issue virtually identical to what is described here:

[RP2040 with USB host no mount callback tuh_hid_mount_cb](https://forums.adafruit.com/viewtopic.php?t=204637)

Having tried all manner of library versions I could conceive, not once have I ever been able to trigger invocation of `tuh_mount_cb`. And just as described in the thread, `tuh_umount_cb` is invoked, but only once. Plugging and unplugging the USB device does not result in further calls to `tuh_umount_cb`.

In the case of pre-compiled `uf2` files, I have tested both the factory reset file shared on the [Adafruit page](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/factory-reset-2), as well as the `device_info.ino.uf2` posted in the forum thread linked above. Both operate as expected, displaying the information of the keyboard that is attached. As requested in the thread, I soldered pins to the RX/TX and dumped the debug logs. They are attached to this report.

Needless to say, this issue is quite strange, and it makes me wonder if there is some subtle overflow issue that is only manifesting with certain memory layouts. If there is anything I can test to help resolve this issue, please let me know.

### How to reproduce ?

1. Load simple device_info sample.
2. Compile/upload to board.
3. Observe errors described.

### Debug Log

[TinyUsbDebug.log](https://github.com/user-attachments/files/19660348/TinyUsbDebug.log)

### Screenshots

Tools configuration:

[image]

hathach, 2025-04-09:
Seem like TinyUSB having difficult getting device descriptor. There is possibly some issue with tinyusb + pio-usb, we have fixed some of the issues recently but probably not all, particularly with low speed device such as mouse/keyboard. Do you have any other devices to test with, e.g a Fullspeed mouse/keyboard or thumbdrive etc.. If it only happen with a few specific device, it would be helpful if you could provide the device detail as well make/model and/or its descritpor (getting from PC).

John-Leitch, 2025-04-09:
Thanks for the quick reply @hathach. It happens with every device I have tested, but I'm not sure if any were fullspeed. Later today I will test with a fullspeed device and provide the debug output.

John-Leitch, 2025-04-09:
@hathach, logs from a full speed device attached. Looks to be close to the same. Given that the uf2's built by others work, I may investigate compiling on Linux to see if there is any difference. However, even if that does work, I still think this is cause for concern.

[TinyUsbDebug_FullSpeed.log](https://github.com/user-attachments/files/19675452/TinyUsbDebug_FullSpeed.log)

John-Leitch, 2025-04-10:
Just tested on a Debian 12 VM using Arduino 1.8.19. Same results, unfortunately.

John-Leitch, 2025-04-10:
@hathach, I have some bittersweet news. I have gotten the device_info example working:

```
Device attached, address = 1
Device 1: ID 1c4f:0002
Device Descriptor:
  bLength             18
  bDescriptorType     1
  bcdUSB              0110
  bDeviceClass        0
  bDeviceSubClass     0
  bDeviceProtocol     0
  bMaxPacketSize0     8
  idVendor            0x1c4f
  idProduct           0x0002
  bcdDevice           0330
  iManufacturer       1     SIGMACHIP
  iProduct            2     USB Keyboard
  iSerialNumber       0     
  bNumConfigurations  1
Device 1: ID 1c4f:0002 SIGMACHIP USB Keyboard
```

However, the means by which I did so are cause for concern: the example began working when I changed the optimization settings from `-Os` to `-O0`. Clearly, this should not affect the outcome of the test. This seems to further indicate memory corruption or perhaps some form of overread, only causing failures with certain memory layouts.

Attached is the USB debug output from the same half-speed keyboard that generated the failure logs. Perhaps the difference will help pin the issue down.

hathach, 2025-04-10:
it is probably a race condtion, since I am working on rp2040, I will try to troubleshoot this as well.

John-Leitch, 2025-04-11:
That would make sense as well. Working my way back from `tuh_umount_cb`, I'm trying to see if I can determine why it is only called once. Adding some additional logging to `process_removing_device`:

```cpp
      TU_LOG_USBH(
        "process_removing_device() checking device id %u\r\n"
        "  rhport:   %u\r\n"
        "  hub_addr: %u\r\n"
        "  hub_port: %u\r\n"
        "  speed:    %u\r\n",
        dev_id,
        dev->rhport,
        dev->hub_addr,
        dev->hub_port,
        dev->speed);
```

I see that `process_removing_device` is indeed called every time a device is unplugged. The first time this occurs, a device with `rhport` of `1` is present in the array:

```
[1:0:0] process_removing_device() called
process_removing_device() checking device id 0
  rhport:   1
  hub_addr: 0
  hub_port: 0
  speed:    1
```

Thus the callback is invoked. However, every time thereafter, all the devices are zeroed:

```
[1:0:0] USBH DEVICE REMOVED
[1:0:0] process_removing_device() called
process_removing_device() checking device id 0
  rhport:   0
  hub_addr: 0
  hub_port: 0
  speed:    0
process_removing_device() checking device id 1
  rhport:   0
  hub_addr: 0
  hub_port: 0
  speed:    0
process_removing_device() checking device id 2
  rhport:   0
  hub_addr: 0
  hub_port: 0
  speed:    0
process_removing_device() checking device id 3
  rhport:   0
  hub_addr: 0
  hub_port: 0
  speed:    0
process_removing_device() checking device id 4
  rhport:   0
  hub_addr: 0
  hub_port: 0
  speed:    0
```

So no callback. Not sure if this provides any clues as to what may be occurring, but I hope to continue picking away at this as time permits.

hathach, 2025-04-11:
this should possibly fixed the issue https://github.com/sekigon-gonnoc/Pico-PIO-USB/pull/176, would you mind giving it a try ? you need to checkout/replace your installed pio-usb

John-Leitch, 2025-04-11:
@hathach, awesome, device_info reliably works now. I hope to do some further testing over the window and I will let you know if I hit any issues.

hathach, 2025-04-14:
thank you for the update, let wait for the PR to be merged and got release then we can close this one.

hathach, 2025-04-14:
This is fixed by pio-usb 0.7.1 thank you for your detail report

closed 2025-04-14 by hathach

weavermedia, 2026-10-03:
Same symptom here on the Feather RP2040 USB Host, and I can add a data point: it's CPU-speed dependent.

- arduino-pico 6.1.1, Adafruit TinyUSB 3.7.7, macOS, arduino-cli 1.5.1
- DualRole/Simple/device_info example, USB Stack = Adafruit TinyUSB

At 120 MHz: host starts, device gets address 1, tuh_mount_cb never fires. On unplug the example logs "Device removed, address = 1". Same with an Arturia MicroLab mk3 (full speed MIDI) and an iLok dongle.

At 240 MHz: both devices enumerate within a second and print their full descriptors. Tested with Pico PIO USB 0.7.1 and 0.7.2, identical results at each clock speed, so the 0.7.1 fix doesn't cover this case.

No other changes between the two builds, just freq=120 vs freq=240.