adafruit/Adafruit_TinyUSB_Arduino #527: Wireless mouse leads to core hang up / How to enable logging

int0h, 2025-05-19:
### Operating System

Linux

### Arduino IDE version

2.3.6

### Board

Adafruit Feather RP2040 with USB Type A Host

### ArduinoCore version

4.5.3 (not sure)

### TinyUSB Library version

3.4.4

### Sketch as ATTACHED TXT

I used some examples from here:  https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/DualRole/HID/hid_mouse_tremor_filter/hid_mouse_tremor_filter.ino

I did slight modifications: I added `HEARTBEAT` logging, but code behaves the same way without this changes, I also tried other examples it always behaves the same.

```cpp
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example demonstrates use of both device and host, where
 * - Device run on native usb controller (roothub port0)
 * - Host depending on MCUs run on either:
 *   - rp2040: bit-banging 2 GPIOs with the help of Pico-PIO-USB library (roothub port1)
 *   - samd21/51, nrf52840, esp32: using MAX3421e controller (host shield)
 *
 * Requirements:
 * - For rp2040:
 *   - [Pico-PIO-USB](https://github.com/sekigon-gonnoc/Pico-PIO-USB) library
 *   - 2 consecutive GPIOs: D+ is defined by PIN_USB_HOST_DP, D- = D+ +1
 *   - Provide VBus (5v) and GND for peripheral
 *   - CPU Speed must be either 120 or 240 Mhz. Selected via "Menu -> CPU Speed"
 * - For samd21/51, nrf52840, esp32:
 *   - Additional MAX2341e USB Host shield or featherwing is required
 *   - SPI instance, CS pin, INT pin are correctly configured in usbh_helper.h
 */

/* Example sketch receive mouse report from host interface (from e.g consumer mouse)
 * and apply a butterworth low pass filter with a specific CUTOFF_FREQUENCY on hid mouse movement report.
 * Filtered report are send via device interface (to PC) acting as a "Mouse Tremor Filter".
 */

// USBHost is defined in usbh_helper.h

#define CFG_TUSB_DEBUG 3
#define CFG_TUSB_DEBUG_PRINTF(...) Serial.printf(__VA_ARGS__)

#include <Adafruit_TinyUSB.h>
#include <pio_usb.h>
#include "usbh_helper.h"

#define TU_LOG(...)   Serial.printf(__VA_ARGS__)
[... 289 lines cut ...]
```
It might have had a good effect (mouse was responsive for longer period of time, around 5 minutes instead of 2), but it might have been an measurement error. (these hangups feel really random and not consistent)

All mice work without any issues with regular PC's of course.

### How to reproduce ?

1. take any (at least all I tried) host usb sample code
2. connect wireless (or maybe this specific model) mouse
3. wait
4. it stops working

### Debug Log

I actually could not enable TinyUSB internal logging, it possibly would really helpful to see if there are any messages leading to the issue. **I would appreciate any advice on how to enable the logs.** 

I use Arduino IDE 2.3.6 on Linux + Adafruit board and Adafruit_TinyUSB, I tried editing `~/.arduino15/packages/rp2040/hardware/rp2040/4.5.3/platform.txt`, and adding `CFG_TUSB_DEBUG=3` : 
```
compiler.c.extra_flags=-DCFG_TUSB_DEBUG=3
compiler.cpp.extra_flags=-DCFG_TUSB_DEBUG=3
```

I tried `#define CFG_TUSB_DEBUG=3` at the top of the sketch source code and some other things but I cannot see any logs. I also noticed that TinyUSB uses Serial1 for logging (I might be completely wrong though) and I tried `SERIAL_TUSB_DEBUG=Serial` but it did not work neither. 

So if someone can help me turn with that it would be great.

hathach, 2025-05-20:
There is a few update to pio-usb which isn't released yet, make sure
- use the git latest on both pio-usb and this repo.
- after enabed -DCFG_TUSB_DEBUG=2, the log will be output via UART tx/rx. Try to hook an ftdi/cp210x to those pin and use different terminal. USB log cannot be printed to cdc since that is also usb communication, which will just loop forever.

Edw1n-G, 2025-05-21:
Hi, I’m experiencing a very similar issue where the core hangs after a few minutes.
Setup:
-     Wired mouse
-     dual_role hid_mouse_log_filter.ino example (unchanged)
-     TinyUSB version 3.5.0
-     Pico-PIO-USB version 0.7.2
-     Same board and IDE

Observation:
    Without moving the mouse, the core hangs after about 15 minutes.
    When moving the mouse:

        ~5 minutes with Boot Protocol (3 bytes)
        1–3 minutes with Report Protocol (11 bytes) added tuh_hid_set_protocol for testing

I'm just getting started with this board and TinyUSB, so I’m afraid that’s all I can contribute at the moment.

int0h, 2025-05-25:
Update: 
- I bought a USB UART adapter connected it to the board and managed to see some logs
- I used `0.7.3` (which was released 3 days ago) and latest (commit hash: `805aa2d41ad9e607c1d086fb7bd197e5a1dd88e2`) from this repo

Unfortunately it did not help :( 

The logs are quite verbose (with `CFG_TUSB_DEBUG=3`) but I could not see anything wrong there, no errors, no warnings, no weird messages. All seem to work just fine until it stops. 

The fact that it happens so inconsistently and I cannot correlate it to anything makes the troubleshooting process quite uncomfortable.

If anyone has any idea of how it can be fixed please let me know, I'm planning to try another board to see if it's gonna be any better there.

Mascota404, 2025-06-09:
what kinda fixes is it forcing it to poll every 1ms like this:
```
  static unsigned long lastUsbTaskTime = 0;

 unsigned long currentMicros = micros();

  if (currentMicros - lastUsbTaskTime >= 1000) {
    USBHost.task();    
   tight_loop_contents(); 

  lastUsbTaskTime = currentMicros;
  }    
```  
But there is also a really weird sensitivity issue that happens, where if you go to a mouse polling test website and just draw circles with your mouse, every 20 seconds you lose like 100 hz and it gets a little slower before it recovers. I've been looking into this issue for a while, so if you want contact me on discord: mascotadev

Mascota404, 2025-06-19:
Using semaphores from FreeRTOS fixes the crashing entirely. Like this:

```
#include <memory>

#include "hid_report_parser.h"

SemaphoreHandle_t usbSemaphore;

// pio-usb is required for rp2040 host
#include "pio_usb.h"
#include "pio-usb-host-pins.h"
#include "Adafruit_TinyUSB.h"

#define CFG_TUSB_RHPORT0_MODE OPT_MODE_HOST
#define CFG_TUSB_OS OPT_OS_FREERTOS 

#include <FreeRTOS.h>
#include <semphr.h>

#include "pico/util/datetime.h"

void setup() {

  mouse_hid.setPollInterval(1);
  mouse_hid.begin();

  

  while (!USBDevice.mounted()) delay(1);

  usbSemaphore = xSemaphoreCreateBinary();
  xSemaphoreGive(usbSemaphore);

}

if (xSemaphoreTake(usbSemaphore, pdMS_TO_TICKS(1))) {
    current_buttons = in_buttons;

    int8_t  in_wheel = m.axes[hid::MouseConfig::V_SCROLL];

   
   int16_t dx = m.axes[hid::MouseConfig::X];
    int16_t dy = m.axes[hid::MouseConfig::Y];

    uint8_t out[7] = {
     in_buttons,                     // 1 byte buttons
    (uint8_t)(dx & 0xFF),           // X low byte
    (uint8_t)((dx >> 8) & 0xFF),    // X high byte
    (uint8_t)(dy & 0xFF),           // Y low byte
    (uint8_t)((dy >> 8) & 0xFF),    // Y high byte
    (uint8_t)in_wheel,              // 1 byte vertical wheel
    (uint8_t)0                 // 1 byte horizontal wheel (pan)
   };

  

   if (mouse_hid.ready()) {
    mouse_hid.sendReport(0, out, sizeof(out));
    }
    xSemaphoreGive(usbSemaphore);

 }

```

hqo998, 2025-08-30:
> Using semaphores from FreeRTOS fixes the crashing entirely. Like this:
>
> ```
> #include <memory>
>
>
> #include "hid_report_parser.h"
>
> SemaphoreHandle_t usbSemaphore;
>
>
>
> // pio-usb is required for rp2040 [...]

Cheers helped me with getting me a solution that worked too

Tools  > Operating System > FreeRTOS SMP

`#include "usbh_helper.h"
#include "tusb.h"
#include "Adafruit_TinyUSB.h"
#include "hid_mouse_reports.h"
 
#define CFG_TUSB_RHPORT0_MODE OPT_MODE_HOST
#define CFG_TUSB_OS OPT_OS_FREERTOS
#define __FREERTOS 

#include <FreeRTOS.h>
#include <semphr.h>
SemaphoreHandle_t usbSemaphore;

// Core 0 - device
void setup() {

  if (!TinyUSBDevice.isInitialized()) {
    TinyUSBDevice.begin(0);
  }

  //disable com port
  TinyUSBDevice.clearConfiguration();

  //device info
  TinyUSBDevice.setDeviceVersion(0x0109);
  TinyUSBDevice.setID(0x3367, 0x1964);

  Serial1.begin(115200);

  // Setup HID
  usb_hid.setPollInterval(1);
  usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
  usb_hid.begin();

  // If already enumerated, additional class driverr begin() e.g msc, hid, midi won't take effect until re-enumeration
  if (TinyUSBDevice.mounted()) {
    TinyUSBDevice.detach();
    delay(10);
    TinyUSBDevice.attach();
  }
}

// Core 0 - device
void loop() {

  // Process incoming serial commands
  processSerialCommand();
  
  // Handle PC commands
  if (pc_command_pending) {
    

    // if (xSemaphoreTake(usbSemaphore, pdMS_TO_TICKS(1))) {
      hid_mouse_report_16bit_t rpt = {0};
      rpt.x = pc_dx;
      rpt.y = pc_dy;
      rpt.wheel = pc_wheel;
      rpt.buttons = pc_buttons;
      
      while (!usb_hid.ready()) { yield(); }
      usb_hid.sendReport(0, &rpt, sizeof(rpt));
      
      // For clicks, send release after a short delay
      if (pc_buttons != 0) {
        delay(50);
        rpt.buttons = 0;
        while (!usb_hid.ready()) { yield(); }
        usb_hid.sendReport(0, &rpt, sizeof(rpt));
      }
      
      pc_command_pending = false;

      // xSemaphoreGive(usbSemaphore);
    // }
  }

// Core1 - host
void setup1() {

  usbSemaphore = xSemaphoreCreateBinary();
  xSemaphoreGive(usbSemaphore);  // initially available

  // configure pio-usb: defined in usbh_helper.h
  rp2040_configure_pio_usb();

  // run host stack on controller (rhport) 1
  // Note: For rp2040 pico-pio-usb, calling USBHost.begin() on core1 will have most of the
  // host bit-banging processing works done in core1 to free up core0 for other works
  USBHost.begin(1);
  delay(3000);
  Serial.print("USB D+ Pin:");
  Serial.println(PIN_USB_HOST_DP);
  Serial.print("USB 5V Pin:");
  Serial.println(PIN_5V_EN);
}

// Core1 - host
void loop1() {
  if (xSemaphoreTake(usbSemaphore, pdMS_TO_TICKS(1))) {
    USBHost.task();  // safe access
    xSemaphoreGive(usbSemaphore);
  }
  // Serial.println("loop");

}

`