adafruit/Adafruit_TinyUSB_Arduino #531: ESP32-S3 not being recognized as MIDI device on PC

jhsa, 2025-05-26:
### Operating System

Windows 10

### Arduino IDE version

Arduino IDE 2.3.5

### Board

ESP32-S3 Dev Module

### ArduinoCore version

ESP32 By Espressif Systems 3.2.0

### TinyUSB Library version

Adafruit TinyUSB 3.5.0

### Sketch as ATTACHED TXT

"midi_test.ino" Example

### Compiled Log as ATTACHED TXT

N/A

### What happened ?

Flashed the "midi_test.ino" example but the ESP32-S3 doesn't appear on the PC as a Midi device, but as "COM 17 (ESP32 Family Device).

### How to reproduce ?

Same as explained above..

### Debug Log

N/A

### Screenshots

[image]

hathach, 2025-05-29:
USB CDC on Boot must be Enabled.

jhsa, 2025-05-29:
> USB CDC on Boot must be Enabled, it is mentioned in the readme

I did compile it  with USB CDC on Boot enabled, but I believe that DFU on boot must be disabled as well?
I managed to make it work, well, sort of. There are a couple issues.
The device is recognized and it works on my Android Smartphone, but it is not recognized on my PC with windows 10.
Also, there is another problem when using this library on the ESP32-S3. If we want to have USB Midi and normal serial Midi working together, enabling the USB CDC on boot will prevent the Serial Midi from working and will even make the Board crash. We need to use Serial2 for the Serial Midi and configure pins as TX and RX.  Example below.
Anyway, the tittle of this topic is still valid as TinyUSB is not recognized as a Midi device by windows 10. I still have win7 on another partition and will test. Will report as soon as I have some time. Thank you for all..

```
#include <MIDI.h>
#include <HardwareSerial.h>

HardwareSerial MySerial(2); // define a Serial for UART1
const int MySerialRX = 11;
const int MySerialTX = 12;

MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, MIDI);

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
     Serial2.begin(31250, SERIAL_8N1, MySerialRX, MySerialTX);
    
    MIDI.begin(MIDI_CHANNEL_OMNI); 

     digitalWrite(LED_BUILTIN, HIGH);
    delay(200);
    
      MIDI.sendNoteOn(42, 127, 1);                  // Launch MIDI and listen to channel 4
digitalWrite(LED_BUILTIN, LOW);
}

void loop()
{
    if (MIDI.read())                    // If we have received a message
    {
        digitalWrite(LED_BUILTIN, HIGH);
        MIDI.sendNoteOn(42, 127, 1);    // Send a Note (pitch 42, velo 127 on channel 1)
        delay(1000);		            // Wait for a second
        MIDI.sendNoteOff(42, 0, 1);     // Stop the note
        digitalWrite(LED_BUILTIN, LOW);
    }
}
```

jhsa, 2025-05-29:
I have just tested another project I have built based on a STM32 board with USBComposite library and it is recognized by windows 10. Could the problem be with the TinyUSB driver?

nikthefix, 2025-06-23:
@jhsa I had the same problem and then noticed that my 'device' was being enumerated as ESP-DEV or JTAG.
I removed this device in Device Manager and scanned for changes.
The S3 now enumerated correctly.

It seems that Win10 / 11 hijacked the S3 and failed to release it as the S3 switched from the initial boot condition of JTAG to user defined TinyUSB device type.

Sounds like a timing issue with either TinyUSB or the S3 Core.
But I've not had a problem since removing the device. I only had to do it once.

hathach, 2025-07-17:
closed due to lack of response. Try the update stock midi_test, which include the detach/attach to see if that works for you.

closed 2025-07-17 by hathach

jhsa, 2025-07-20:
Ok, I have updated the TinyUSB library to the latest 3.7.1 (I think), and added the Detach/Attach code to my projects code as seen below. The result is the same as before. Windows 10 does NOT recognize the device. It shows up in "Device manager" under "Ports(COM & LPT)" as " USB Serial Device (COM21)"
It does not show up as Midi Device.
On android, it does show up as Midi device and I can send and receive Midi. But I must connect it to my phone wit the phone unlocked, otherwise it is not recognized.

Here is the code I have mentioned above and the Board settings.

```
//************ USB MIDI init ***************
  TinyUSBDevice.setManufacturerDescriptor("My_Manufacturer");
  TinyUSBDevice.setProductDescriptor("iL9_USB");
  // usb_midi.setStringDescriptor("iL9_USB");

  // Initialize MIDI, and listen to all MIDI channels
  // This will also call usb_midi's begin()
  MidiUsb.begin(MIDI_CHANNEL_OMNI);

  // If already enumerated, additional class driverr begin() e.g msc, hid, midi won't take effect until re-enumeration
  if (TinyUSBDevice.mounted()) {
    TinyUSBDevice.detach();
    delay(10);
    TinyUSBDevice.attach();
  }

  // Attach the handleNoteOn function to the MIDI Library. It will
  // be called whenever the Bluefruit receives MIDI Note On messages.
  MidiUsb.setHandleProgramChange(USB_ProgramChange);
  MidiUsb.setHandleControlChange(USB_ControlChange);
```

[image]

nikthefix, 2025-07-21:
@jhsa Take a look in device manager (with the S3 board disconnected) and select 'show hidden devices' from the View drop down menu.
Expand 'Universal Serial Bus Devices' 
If you see ESP_DEV or USB JTAG or similar then uninstall the device along with the driver if the option is offered.
Restart PC.

jhsa, 2025-07-21:
> [@jhsa](https://github.com/jhsa) Take a look in device manager (with the S3 board disconnected) and select 'show hidden devices' from the View drop down menu. Expand 'Universal Serial Bus Devices' If you see ESP_DEV or USB JTAG or similar then uninstall the device along with the driver if the option is offered. Restart PC.
> 
> I had the same problem not just with the midi example but with all the examples. In my case it turned out that if I forgot to 'Enable CDC on boot' or set 'USB-OTG (TinyUSB)' or both, from then on all examples would fail to enumerate properly even after I'd corrected the build settings and re-flashed. This behaviour is repeatable for me on two different Win10/11 computers. The stubborn ESP_DEV or USB JTAG entries re-appear if I deliberately misconfigure the build options.

This seems to have worked. I can now send and receive midi on my Windows 10 PC. Thank you so much.

[@hathach](https://github.com/hathach)This explanation should perhaps be shared somewhere on the information page of the library.

nikthefix, 2025-07-21:
@jhsa Glad it worked for you too. At least we have a work around.

jhsa, 2025-07-21:
> [@hathach](https://github.com/hathach) For what it's worth I don't think the problem is with Adafruit_TinyUSB. I use the S2 as much as the S3 and have never had issues with the S2 even when I've inadvertently caused #PID clashes thru reckless experimentation. Win10/11 did not exhibit a problem with either device until fairly recent updates of both. My Linux machines also have no problem. I've also not had any problems anywhere since ESP_Arduino 3.2.1

Oh, I have just noticed I still have 3.2.0 ESP Boards version.. Will update shortly. Thank you.

jhsa, 2025-07-28:
> [@jhsa](https://github.com/jhsa) let me know how is your testing with latest version. We can re-open this issue if needed.

Yeah, it is still working, thank you. I have the last version installed. I also followed the workaround mentioned before, and that made it work.
after that I already compiled different codes with different CDC options, and it seems my USB projects are still recognized on my windows 10 PC. I have another PC that never ran Midi that I could also test.
Thank you so much for looking into this. Amazing library... 
Now if we could make USB MIDI Host work, that would be wonderful :D

nikthefix, 2025-07-28:
@hathach @jhsa Tested with ESP V3.3.0 - all working. However, I'm still able to cause the problem if I deliberately set the build options wrongly, recovering only by deleting the JTAG device in Device Manager as outlined above. This will be a gotcha for many I imagine.

jhsa, 2025-07-28:
> [@hathach](https://github.com/hathach) [@jhsa](https://github.com/jhsa) Tested with ESP V3.3.0 - all working. However, I'm still able to cause the problem if I deliberately set the build options [...]

Yeah, I think your solution should be mentioned on the Library's main page.

AtmosphEng, 2025-08-06:
Hi, I also have the same or similar problem. I'm building with PlatformIO on VSCode. I've tried two types of Lilygo boards: t-display-s3 and t-embed-s3. Tried both Windows 10 and Windows 11. In Device Manager I've tried deleting drivers, scanning for new hardware, and rebooting my PC, all as described above. The results are the always the same every time i.e. the programmed esp32-s3 appears as a COM port and as a JTAG device. Any new ideas or tests to run please?

nikthefix, 2025-08-06:
@AtmosphEng When you see the JTAG device in Device Manager, delete the driver as well as the device.
Something else to try is connecting your S3 board via a USB 2 hub to the PC. Someone in another thread found that this introduced a small timing difference at enumeration and cured the problem all the while the hub was used.

AtmosphEng, 2025-08-06:
@nikthefix Thanks for your quick reply. When I 'delete' the driver in Device Manager, I right-click on the 'USB JTAG/serial debug unit' and select  Uninstall device - it only ever gives me the option to uninstall, and never offers me a delete driver option. Do you also delete the COM port which also appears with the JTAG driver? I've tried this also, but had no success.
I've also tried 3x different USB2 hubs as you suggested but again unfortunately no success. Where exactly and how does your MIDI device driver show in Device Manager when it works for you?

nikthefix, 2025-08-07:
@AtmosphEng Ah, for the Adafruit TinyUSB examples to work the  build options should be:

-D ARDUINO_USB_ON_BOOT=1    (CDC on boot enable)
-D ARDUINO_USB_MODE=0          (OTG mode enable)

My usb-midi devices appear in Device manager under 'Sound, video and game controllers'.
In S3 OTG mode you may have to be in bootloader to upload new sketches - hold 'boot' button while tapping 'reset' button.