adafruit/Adafruit_TinyUSB_Arduino #534: hid_generic_inout not compiling platformio, esp32s3

Macka323, 2025-06-06:
### Operating System

Windows 11

### Arduino IDE version

Platformio, 6.1.18

### Board

ESP32-S3

### ArduinoCore version

Core 6.1.18       Home 3.4.4

### TinyUSB Library version

Adafruit TinyUSB Library@^3.6.1

### Sketch as ATTACHED TXT

[main.txt](https://github.com/user-attachments/files/20635441/main.txt)

### Compiled Log as ATTACHED TXT

[compile.txt](https://github.com/user-attachments/files/20635442/compile.txt)

### What happened ?

When i try to compile i get the error that there are multiple defentions of functions (i am not sure, i think that is the problem from the compile log)

### How to reproduce ?

Create new platformio project.
Add adafruit/Adafruit TinyUSB Library@^3.6.1 as a library.
Change main.cpp  with the example from above or the hid_generic_inout example.
This is what i use in `platformio.ini` in order to enable usb.
```ini
[env:esp32s3]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
lib_deps = adafruit/Adafruit TinyUSB Library@^3.6.1
build_unflags = 
	-DARDUINO_USB_MODE=1
build_flags = 
	-DARDUINO_USB_MODE=0
	-DARDUINO_USB_CDC_ON_BOOT=1
```

Try to build.

hathach, 2025-07-05:
1. please provide exact arduino-esp32 core release vesrion
2. try to see if compiling using Ardino IDE with latest arduino-ep32 work with latest TinyUSB. If it works, pio compiler may miss something, please try to use the raw compile log to force that on pio compiler.

StarWitch, 2025-07-14:
I'm also having this problem, with the exact same "multiple definitions" error. It seems to be a conflict with the TinyUSB built into the espressif framework.

```
PLATFORM: Espressif 32 (6.11.0) > Unexpected Maker FeatherS3 Neo
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

hathach, 2025-07-15:
@StarWitch what is your arduino-esp32 version ? if it is not 3.2 then try with 3.2